queens-satellite-team/mock-sat #24: Get single axis PI control loop working with maxon motor

WoJiaoCal99, 2020-07-16:


WoJiaoCal99, 2020-07-19:
Good [article ](https://www.maxongroup.co.uk/maxon/view/news/How-to-configure-a-maxon-ESCON-Speed-Controller) for basics of configuring maxon controller

WoJiaoCal99, 2020-07-19:
Motor [Datasheet ](http://edge.rit.edu/edge/P13021/public/DDR/Maxon%20Generator%2030W.pdf)

WoJiaoCal99, 2020-07-19:
[Page](https://www.maxongroup.com/maxon/view/content/escon-detailsite) for download of maxon controller setup software

anthonysteel, 2020-07-20:
Looks good.

WoJiaoCal99, 2020-07-23:
The [controller](https://www.maxongroup.com/maxon/view/product/control/4-Q-Servokontroller/466023) requires multiple inputs as specific signals:
-desired angular velocity (analog or digital PWM)
-angular acceleration (analog)
-direction (digital)
-enable (digital)
-all outputs come as analog
The raspi can only create digital and PWM signals (PWM with limited accuracy), thus cannot set angular acceleration. Simplest solution is probably an i2c slave to read and write analog signals and relay back to raspi.

WoJiaoCal99, 2020-07-23:
The motor connects via flex-cable (FPC cable) and needs an adapter for prototyping. [This 11-pin adapter](https://www.digikey.ca/en/products/detail/1-1734248-1/A101417CT-ND/2567553?itemSeq=333183093) should have the right spacing for the 5 normal width connections, and 3 wide (motor winding) connections on the flex cable.

WoJiaoCal99, 2020-08-06:
A few developments for maxon motor control:
-Previous adapter is a surface mount connector for PCBs. [This version](https://www.digikey.ca/en/products/detail/te-connectivity-amp-connectors/1-84982-1/1893492) should have enough space for soldering and prototyping. 
-[This DAC](https://www.digikey.ca/en/products/detail/maxim-integrated/MAX11300PMB1%23/4914312?utm_adgroup=Development%20Boards%2C%20Kits%2C%20Programmers&utm_source=google&utm_medium=cpc&utm_campaign=Shopping_Supplier_Maxim%20Integrated_0175_Co-op&utm_term=&productid=4914312&gclid=EAIaIQobChMIrLfLyfDj6gIVAoTICh3JWAINEAQYASABEgLbFvD_BwE) currently only works with the [Munich GUI Software](https://www.maximintegrated.com/en/design/software-description.html/swpart=SFW0004100C) on a laptop. [This library](https://os.mbed.com/teams/MaximIntegrated/code/MAX113XX_Pixi/shortlog/) seemed promising but hasn't worked with the raspi thus far. It should be noted that the DAC can be used as an oscilloscope and analog/digital signal IO and would be useful to have working with either raspi/arduino for both prototyping and potentially sensor/motor control in the final version of the moc-sat.
-Moving forwards with the [Sparkfun MCP4725 DAC](https://www.sparkfun.com/products/12918). If we can create our own i2c communication protocol with the chip, we could print our own on a custom PCB based on the [public design](https://github.com/sparkfun/MCP4725_Breakout/tree/v14).
-Moving from raspi to arduino for compatibility with the new DAC, as well as future development including RTOS and integration with the OBC subteam.

WoJiaoCal99, 2020-08-14:
Motor is now working:
-acc and vel are controlled by two MCP4725 DACs 
-second flex-cable connection can be soldered for motor connections
-All i2c connections have been tested with arduino's [freeRTOS](https://www.freertos.org/a00110.html) and work - see commit 09b34a6a7239b34cd3f976704b3c3b82ddb7321e
-Moving forwards by creating control loop function to be run by the RTOS