queens-satellite-team/mock-sat #28: Send image from Raspberry Pi SD card with NRF24L01 Transceiver

JoshW7, 2020-07-18:
I got an Arduino nano with a NRF24L01 set up to work from the Raspberry pi. The nano couldn't communicate with the R Pi using the default linux IDE so I installed the ARM version of Arduino 1.8.0 following this [forum thread](https://www.raspberrypi.org/forums/viewtopic.php?t=170210).   

I installed the same nRF24 library on the pi and successfully sent serial data to my laptop using the transceivers.

Once we get [lossless data transmission](https://github.com/queens-satellite-team/moc_sat/issues/26) working, it should be easy to set up on the pi.

anthonysteel, 2020-07-20:
Another option for people working on this if they have a problem with flashing their Arduino using the Raspberry Pi: you can flash the Arduino using any computer or IDE that supports it and then just connect it to the RP, using the RP only for non-firmware development. Assuming that the Arduino code is relatively stable and won't need to be updated frequently,  this is a reasonable option.

I agree once, we get the lossless data transmission working it should carry over with no problems.

JoshW7, 2020-07-22:
RF24 has a built in acknowledgement (ACK) feature that allows for data packets to be sent without data loss. If you have the Arduino RF24 library installed, you can look at some of the ACK example sketches. 

This is essentially a built in system for bi-directional communications and I believe it's what we're looking for. I'm going to mess around with sending huge data streams this way and see how fast/lossless it is. If all goes well, I'll try it with an image.

anthonysteel, 2020-07-22:
Perfect.

anthonysteel, 2020-07-22:
I’m going to close out #26 and we can re-open it if the built in R24 functionality fails.