queens-satellite-team/mock-sat #30: Order parts for EPS

anthonysteel, 2020-07-29:
Decide on what tutorial we are going to follow, and what parts are required. Write a BOM and push it to MocSat docs.

anthonysteel, 2020-07-29:
@varnikaagupta Was [this](https://www.digikey.ca/en/maker/projects/charge-batteries-with-an-arduino/b9d6ff74ecea437da8126d097d680d56) the tutorial Josh was talking about? You will also want to order a soldering iron, solder, rosin flux, solder remover and/or solder wick.

varnikaagupta, 2020-07-30:
Yep, I'll go through and get down all the equipment needed

varnikaagupta, 2020-07-30:
> Decide on what tutorial we are going to follow, and what parts are required. Write a BOM and push it to MocSat docs.

@anthonysteel Since we want to incorporate solar charging for the moc sat, would it be better to start directly with a tutorial like this? https://create.arduino.cc/projecthub/igorF2/solar-charged-battery-powered-arduino-uno-645d89
What voltage solar cells do we currently have?

anthonysteel, 2020-08-11:
Sorry for the late response to this. I don't know which solar cells are in the room, @JoshW7 might know, but I wouldn't worry about using the proper panels. That tutorial looks good, and yes I would say it is preferable to use panels directly.

You will also want to look at what methods there are for a controlled discharge of the batteries when you are testing the circuit. It might be something as simple as connecting them to a motor or light.